bitcraze/lps-node-firmware #9: Compiling Problem with LPS Node Firmware after Update ALL

ManuelGraziosi, 2016-12-01:
Trying to solve communication's problem with drone and ROS Jade System, i made an Update All Project.
Now it can not compile the Lps Node Firmware (neither by console or IDE Eclipse).
The compile's output is:

> 11:27:02 **** Incremental Build of configuration Default for project lps-node-firmware ****
make all 
python2 tools/make/check-for-submodules.py
arm-none-eabi-gcc -o bin/lps-node-firmware.elf lib/freertos/src/list.o lib/freertos/src/queue.o lib/freertos/src/timers.o lib/freertos/src/tasks.o lib/freertos/src/port.o lib/freertos/src/event_groups.o src/f0/startup_stm32f072xb.o src/f0/system_stm32f0xx.o src/f0/stm32f0xx_it.o src/f0/stm32f0xx_hal_msp.o src/f0/gpio.o src/f0/i2c.o src/f0/spi.o src/f0/system.o src/f0/usart.o src/f0/usbd_conf.o src/eeprom.o src/main.o src/usb_device.o src/usbd_cdc_if.o src/usbd_desc.o src/lps25h.o src/led.o src/cfg.o src/usbcomm.o src/test_support.o src/production_test.o src/uwb.o src/uwb_twr_anchor.o src/uwb_sniffer.o src/uwb_twr_tag.o src/uwb_tdoa_anchor.o hal/stm32f0xx/Src/stm32f0xx_hal_i2c_ex.o hal/stm32f0xx/Src/stm32f0xx_hal_gpio.o hal/stm32f0xx/Src/stm32f0xx_hal_rcc.o hal/stm32f0xx/Src/stm32f0xx_hal_cortex.o hal/stm32f0xx/Src/stm32f0xx_hal_i2c.o hal/stm32f0xx/Src/stm32f0xx_hal_pcd.o hal/stm32f0xx/Src/stm32f0xx_hal_dma.o hal/stm32f0xx/Src/stm32f0xx_hal_pcd_ex.o hal/stm32f0xx/Src/stm32f0xx_hal_rcc_ex.o hal/stm32f0xx/Src/stm32f0xx_hal_spi.o hal/stm32f0xx/Src/stm32f0xx_hal_uart.o hal/stm32f0xx/Src/stm32f0xx_hal.o Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.o Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.o Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ioreq.o Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.o vendor/libdw1000/src/libdw1000.o vendor/libdw1000/src/libdw1000Spi.o src/dwOps.o -mthumb -mcpu=cortex-m0 -DHSI48_VALUE="((uint32_t)48000000)" -DSTM32F072xB --specs=nano.specs --specs=nosys.specs -Ttools/make/stm32f072.ld -lm -lc -u _printf_float -Wl,-Map=bin/.map,--cref,--gc-sections
src/uwb.o: In function `uwbInit':
/home/bitcraze/projects/lps-node-firmware//src/uwb.c:124: undefined reference to `dwAttachReceiveFailedHandler'
src/uwb_tdoa_anchor.o: In function `handleRxPacket':
/home/bitcraze/projects/lps-node-firmware//src/uwb_tdoa_anchor.c:171: undefined reference to `dwGetRawReceiveTimestamp'
src/uwb_tdoa_anchor.o: In function `setupTx':
/home/bitcraze/projects/lps-node-firmware//src/uwb_tdoa_anchor.c:249: undefined reference to `dwSetTxRxTime'
src/uwb_tdoa_anchor.o: In function `setupRx':
/home/bitcraze/projects/lps-node-firmware//src/uwb_tdoa_anchor.c:211: undefined reference to `dwSetTxRxTime'
collect2: error: ld returned 1 exit status
make: *** [bin/lps-node-firmware.elf] Error 1
11:27:08 Build Finished (took 5s.526ms)

krichardsson, 2016-12-01:
Works on travis and my machine. Have you tried to update the git submodules?
```git submodule update```

ManuelGraziosi, 2016-12-01:
yes, I have done the `git submodule init` and then `git submodule update` but no change happened.
I tried to repeat all the step suggested in any procedure.
yesterday I started to have problem with the Ros System (Error: no transform [crazyflie/base_link] to [world]); the last time that I receive this problem I solved it updating the anchor. For this reason I tried to compile the firmware but without success. I'm working with only 1 drone so the anchor are configured in TWR mode.

krichardsson, 2016-12-02:
The undefined references are from the vendor/libdw1000 sub module. The references were added in commit bitcraze/libdw1000@55f86179c7f4ea821130603f4186f80346779700 so as you are missing them it looks as you don't have the right version of the submodule for some reason.

There are a few things you could try:
1. Delete the sub modules and update them
```
rm -r vendor/*
git submodule update
```
2. Clone the full repo
```
git clone --recursive https://github.com/bitcraze/lps-node-firmware.git
```

ManuelGraziosi, 2016-12-05:
Thanks a lot @krichardsson , with your procedure (I followed the step 1) now I can compile without problem. I will soon update all the anchors and I hope that ROS return to work properly.
Thanks again for the support :)

closed 2016-12-05 by ManuelGraziosi

krichardsson, 2016-12-05:
Great!