bitcraze/lps-node-firmware #16: setDelay() AntennaDelay isn't updated.

mikucukyilmaz, 2017-08-10:
Hi , 
When using SetDelay library use dev->antennaDelay to calculate futureTime. But dev->antennaDelay set to 0 in uwbInit.So it doesn't add antennaDelay to futureTime calculation.
```
  dwTime_t delay = {.full = 0};
  dwSetAntenaDelay(dwm, delay);
```
 But in dwInit()
` writeValueToBytes(dev->antennaDelay.raw, 16384, LEN_STAMP);`

ANTENNA_DELAY/2 value in twr_tag & anchor ~16475. I guess in uwbInit() if antennaDelay set correctly it could be used for calculations and
```

#define ANTENNA_OFFSET 154.6   // In meter
#define ANTENNA_DELAY  (ANTENNA_OFFSET*499.2e6*128)/299792458.0 // In radio tick 
```
part of the code can be moved to uwb.h . dev->antennaDelay can bu used for adding & subtracting from departure & arrival times.

ataffanel, 2017-08-21:
Hi,
The antenna delay, when set in the dw1000, is an offset that will be applied to timestamps by the dw1000. We have started setting it to 0 in the dw1000 and applying the offset in sofware instead. The antenna delay in dwInit() is a default value, the software should set its own value.

What problem are you observing?

mikucukyilmaz, 2017-08-21:
Hi,
Actually there is no big problem. In dwSetDelay function when calculating futureTime it looks like it adds to antennaDelay to futureTime but dev->antennaDelay.full is set to 0 so it doesn't add in dwSetDelay method. It looks more appropriate to me using dev->antennaDelay in calculations to make library more understandable.

https://github.com/thotro/arduino-dw1000/pull/226

 I guess this is not the place but i tried this fix and it really improves the precision if you want to check.
Thank you.

closed 2018-01-26 by mikucukyilmaz