bitcraze/lps-node-firmware #26: Question about the TX timestamp calculation

jpjporto, 2018-02-16:
Hi all,

I had a question about the transmit time calculation on the TDOA implementation ([L156-170](https://github.com/bitcraze/lps-node-firmware/blob/37c150ee2575b41f1833eea477505648fe967f6d/src/uwb_tdoa_anchor2.c#L156-L170)).
On the code, you add the TDMA_GUARD_LENGTH and PREAMBLE_LENGTH to the transmit time, however, according to the DW1000 user manual (pg. 26), it looks like the chip already take into account the preamble length when setting the delayed transmit. Similarly, the RX timestamp also corresponds to the start of the PHR marker.

I'm not sure if this actually makes a difference in the end, since the values added are constant and we take the difference between two timestamps. I've been meaning to check this with Vicon, but I haven't had much time lately.

Thanks,
Joao

ataffanel, 2018-02-16:
Hi,
The reason the preamble length is added is to start the actual radio transition in the TDMA time-slot, not before. If we where setting a TX time exactly at the beginning of the time-slot my understanding of the documentation is that the DW1000 will start transmitting the preamble before the time-slot starts so that the PHR marker matches the start of the time-slot.

Making sure there is not transition of an anchor outside its time-slot simplifies the receiving part since the receiver does not have to know the preamble size, it can start the receiver right at the time the timeslot starts.

jpjporto, 2018-02-23:
Thank you, I understand the issue now.

closed 2018-02-23 by jpjporto