bitcraze/lps-node-firmware #27: Implement TDoA anchor protocol 3.0

krichardsson, 2018-05-04:
TDoA 3 will support random tx times and dynamic number of anchors. The main features that we intend to add with this work is:
1. Remove the need for a master anchor. This will make the system more robust since anchor 0 will no longer be a single point of failure.
1. Support more than 8 anchors to be able to cover a larger area

NicksonYap, 2018-05-04:
Just a quick question,
How many Crazyflies can TDoA or TWR support?

Can it go up to 100 or more?

Ideally we'd want it to go up to 1k Crazyflies for future possibility

krichardsson, 2018-05-07:
TWR - one, TDoA - unlimited. Please see https://www.bitcraze.io/loco-pos-system/.
Also, please ask this type of questions in the https://forum.bitcraze.io/ instead, we try to use issues on github for bugs and features.

NicksonYap, 2018-05-07:
@krichardsson  
Thanks for the reply, just found out about the forum.

Good to hear TDoA is theoretically unlimited.

Though I should be asking in the forum, but it seems TWR should support more than 1 device.
Since TDMA is implemented: https://wiki.bitcraze.io/doc:lps:twr-tdma

I have just bought 6 Anchors and 2 crazyflies for TWR mode.
It should work for two Crazyflies, no?

Thanks!

krichardsson, 2018-05-08:
Yes, TWR-TDMA supports 2 CFs. You can read more at https://wiki.bitcraze.io/doc:lps:twr-tdma

xingqing6210, 2018-05-09:
if every anchor will send packet randomly, how to calculate the position by tag?

krichardsson, 2018-05-11:
@xingqing6210 The TX time (in the remote anchor clock) is part of the data in each packet. This is also true for TDoA2 and there is not really any difference in the position estimation functionality in TDoA3.

xingqing6210, 2018-05-12:
one anchor still receive another anchor packet for sync ?

krichardsson, 2018-05-15:
@xingqing6210 Yes, all anchors are listening to the anchors around them

xingqing6210, 2018-05-17:
the anchor send packet's frequence is ?  100ms or others?

krichardsson, 2018-05-17:
@xingqing6210 Experiments showed that packet loss slowly rise with increasing total system packet rate and we loose around 15%@400 packets/s. With higher packet rates the losses increase rapidly so 400 packets/s seems to be a good total rate.
What I have tried to do is to let the anchors decide which transmit rate to use and reduce the rate if there are many other anchors in the system around them. Currently the transmit frequency is throttled based on how many other anchors that an anchor receives messages from, more anchors => lower rate. The TX rate is 50 packets/s per anchor up to 8 anchors, above that the rate is reduced to target 400 packets/s total rate. 16 anchors should end up at about 25 packet/s per anchor.

krichardsson, 2018-05-21:
@xingqing6210 The idea is to try to keep the frequency of packets that tags can use fairly constant throughout the system. 
50 * 8 = 400 Hz, 
25 * 16 = 400 Hz
Tags can use a messages from any anchor to make the TDoA calculation.
This functionality is mainly interesting in large systems where each tag only covers a part of the full system.

xingqing6210, 2018-05-25:
@krichardsson  maybe support tdma tof method for more zone ?

krichardsson, 2018-05-28:
@xingqing6210 I'm not sure what you mean. Could you elaborate a bit?

xingqing6210, 2018-05-29:
@krichardsson  tdoa3 support more two zone, maybe support tdma tof ?

krichardsson, 2018-05-29:
Basic TDoA 3 functionality is implemented. We will consider this as a baseline and close this issue. Further improvements/fixes will use separate issues to the changes the changes more clearly.

closed 2018-05-29 by krichardsson