bitcraze/lps-node-firmware #54: Not able to configure anchor positions

easy98, 2020-11-25:
Hello,

I am not able to configure the anchor positions since when I click the button "get from anchors", no response. I think the connection between the cfclient and crazyflie should be fine. I am so confused about this problem. 

Thanks in advance.
[image]

krichardsson, 2020-11-25:
Hi!

This happens if the anchors do not yet have a position stored in them. If this is the case you have to add the anchors manually using the "Add anchor" button in the "Anchor position configuration" dialog box. 
When you have written positions to the anchor the "get from anchors" function should work.

krichardsson, 2020-11-25:
Related to bitcraze/crazyflie-clients-python#457

easy98, 2020-12-01:
Thanks for quick reply,

So it seems I need to measure the anchor positions manually?

krichardsson, 2020-12-02:
Yes, that is correct. Anchor positions must be measured manually.
We have done some experiments with automated position estimation of anchors but have not finalized it in the system.

easy98, 2020-12-02:
OK, thanks very much!

closed 2020-12-02 by easy98