bitcraze/lps-node-firmware #57: WIRELESS SYNCHRONIZATION OF ANCHOR NODES.

manhtran98bg, 2022-01-19:
Hi all !
I am doing a project using dwm1001 module (with dw1000 uwb core). I want to implement RTLS with this module using TDoA topology.
I have read dw1000 datasheet, it said that dw1000 support both TWR and TDoA schemes. The problem when using TDoA scheme is Synchronization between anchor nodes. This can archive by using wired (SYNC pin in dw1000 and External Clock supplied on EXTCLK pin) or wireless (using UWB to Synchronize). So may i need implement Synchronization  before using your lib (uwb_tdoa_anchor2.c). Please help me ! Thank you very much!

krichardsson, 2022-01-19:
Hi!
We recover the clock drift of the remote anchor by using the time stamps in the received packets. You can find more information here https://www.bitcraze.io/documentation/repository/lps-node-firmware/master/

This repository contains the code for our anchors and they are essentially doing TWR with each other, so that is what you will find here. 

The code that is doing the actual TDoA work is located in the https://github.com/bitcraze/crazyflie-firmware repository. Starting in https://github.com/bitcraze/crazyflie-firmware/blob/2021.06/src/deck/drivers/src/lpsTdoa2Tag.c, but main part of functionality is in https://github.com/bitcraze/crazyflie-firmware/tree/2021.06/src/utils/src/tdoa.

I would suggest to use TDoA3 instead of TDoA2 as it is more robust and flexible.

manhtran98bg, 2022-02-09:
> Hi! We recover the clock drift of the remote anchor by using the time stamps in the received packets. You can find more information here [...]

manhtran98bg, 2022-02-14:
Hi!
Thanks for answer me. We are currently having a problem that needs your advice. As I understand, LPS using Reserve TDoA that means Anchor is Transmitter (Active) and Tag is a Receiver (Passive Listen). And Tag (here is LocoDeck ) listen packet and send it to locate engine (MCU on Crazyflie) to calculate location. So We have 2 questions need to help:
1. If i use LPS TDoA v2, can i use multiple Tag in my system? 
2. In my system i'm going to built, Anchors are Receiver, and Tags are transmitter. Synchronization between Anchors is necessary, so Can I use TDoA v2 to sync Anchors in my system?
Thank you very much!

krichardsson, 2022-02-14:
Hi!

1. Yes, as the tag is only listening, you can use as many tags as you like (compare to a GPS receiver)
2. There is no functionality in the system for using Anchors as receivers and decks as transmitters. The UWB is identical on both sides so from that point of view it is possible. Keep in mind though that the CPU on the anchors is not as powerful the one on the Crazyflie. 

You can read more here https://www.bitcraze.io/documentation/system/positioning/loco-positioning-system/ https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/loco-positioning-system/ https://www.bitcraze.io/documentation/repository/lps-node-firmware/master/functional-areas/tdoa2-vs-tdoa3/

knmcguire, 2022-03-07:
closed due to inactivity.

Just mind that we mostly use forum.bitcraze.io for support.

closed 2022-03-07 by knmcguire