bitcraze/lps-node-firmware #63: Using LPS Node firmware with ESP32 module combined with DWM1000

krishsankar7, 2023-07-04:
Hi Team,  

Right now I am using Makerfabs ESP32 UWB board for indoor positioning system with TWR mode (using DW1000 library). While using 1 tag, this algorithm works perfectly fine, however, when we go for multiple tags, this TWR mode fails. 

As an alternative method, we are now exploring TDOA algorithms. We came across your work on TDOA3 and like to implement the same with our hardware (ESP32+DWM1000). Kindly confirm if this LPS Node firmware using TDOA3 is compatible with this hardware. 

If so, please share us any documents for the same that clearly explain the workflow for this implementation.

krichardsson, 2023-07-05:
Hi,

> While using 1 tag, this algorithm works perfectly fine, however, when we go for multiple tags, this TWR mode fails.

Correct, that is by design, TWR only supports one tag.

> As an alternative method, we are now exploring TDOA algorithms. We came across your work on TDOA3 and like to implement the same with our hardware (ESP32+DWM1000). Kindly confirm if this LPS Node firmware using TDOA3 is compatible with this hardware.

The code in this repository is not written to be compatible with the ESP32, it was designed for [the Loco nodes from us (Bitcraze)](https://www.bitcraze.io/products/loco-positioning-node/). I have no idea how hard it would be to convert it, I would assume there is a fair amount that can be reused though.

krichardsson, 2023-07-05:
If you have further questions and discussions that is related to support, please post them in [our discussions forum](https://github.com/orgs/bitcraze/discussions)

knmcguire, 2023-12-07:
Closing this issue now since it is not applicable for issues

closed 2023-12-07 by knmcguire